Befaco/Oneiroi #1: CV-Ins are reversed on startup

pssssssst, 2024-07-25:
Hi, i built the DIY version of oneiroi and everything works fine except the behavior of the cv-ins for the parameters.
When i turn on the module everything is reversed.
If i use the "shift + mod amount" button combination and then press "record + random" everything starts to work as expected.
But i can't save that state. 
Holding "mod amount" for 2 seconds seems to save (blinks twice), but after a power cycle the cv-ins are reversed again.

Other than that i love the module. Thanks you for putting this together. I have been having non stop fun with this :)
Bets,
Marc

blindorange, 2024-07-26:
I read this about the issue you describe; 
do the 'Shift+Mod Amt' on all the parameter and set them to 100% (fully clock wise), and now the CV seems to work correctly.

In the manual where it says this (section 14):
Then, adjust the knob you want to be affected by the CV from 0% to 100% (the default is 100% for all primary parameters).

I assume there is a bug in the firmware were the default is not set 100%. (in my case it seemed to be -100% i.e. inverted). Now that I've set it to 100% I can't recreate the inverted behaviour

pssssssst, 2024-07-26:
Yeah, thanks. i figured out how to change it to normal behavior.
But as soon as i turn off the module and turn it back on again the inversion is back.
It doesn't bother me too much, but it would be nice if the saving would include these changes.

Diegoterrores, 2024-08-07:
I'm trying to reproduce this issue on several "fresh" units and cannot reproduce it.
It looks like it is something only happening on some units probably due to a human mistake on the testing/calibration procedure.

@pssssssst you can save CV mapping by holding MOD-AMT Button until it blinks so you do not have to do it every time you switch your module ON.
What serial number does your unit have on the back?

pssssssst, 2024-08-11:
Hi,

thanks for the reply.
My unit is a DIY so it doesn't have a serial number.
I will try the calibration again.

My problem is, that holding MOD-AMT doesn't save the cv-settings. It saves everything else but not the cv-settings.
After a power cycle everything is reversed again.

Also i can't seem to be able to connect my owl board via web midi.
Does it need to be powered over the power bus or is a usb connection enough?

octovolt, 2024-08-16:
@blindorange The CV defaults are here, I believe:

https://github.com/Befaco/Oneiroi/blob/main/Ui.h#L183-L192

They appear to be set at 100%, not -100%.

Maybe the problem is in the calibration procedure?

rootapprox, 2024-08-24:
i have the same issue , i calibrated 2 times .. we really need a new firmware ...

pssssssst, 2024-08-25:
> i have the same issue , i calibrated 2 times .. we really need a new firmware ...

can you connect your owl board to the web interface they provide

rootapprox, 2024-08-25:
yes i finally succeeded to connect the owl board to the web interface , changin the cable im using to a Beatstep Pro one , also try to get it connected through chromium application.. but still im kinda disappointed that i have to always click shift random record + mod atm random record to get the normal behaviour of the unit ...

pssssssst, 2024-08-25:
> yes i finally succeeded to connect the owl board to the web interface , changin the cable im using to a Beatstep Pro one , also try to get it connected through chromium application.. but still im [...]

I'm sure they will fix it.
Maybe i can find another cable and try it with that. Does the board need to be powered seperatly or is usb power enough?

rootapprox, 2024-08-25:
yes the module needs to be powered to be detected i guess from my experience

blindorange, 2024-09-18:
I think I maybe getting confused about the order of calibration, GitHub says power on first then set pots to 12, the manual switches these around, set pots to 12, then power on, not sure which is correct, what are you folks doing?

pssssssst, 2024-10-05:
i tried both. But it doesn't seem to make a difference to me.
I also tried the new beta, but the reversed cv issue is still present

hirnlego, 2024-10-05:
@pssssssst try resetting the CV mapping (RECORD + RANDOM while in CV map) then save and restart. Are you using attenuverters instead of attenuators? The latest beta should resolve the problem of attenuverters values saved incorrectly, but they need to be reset.

pssssssst, 2024-10-05:
I tried that a lot of times. With attenuator and attenuverter mode. But it just doesn't save the configuration.

hirnlego, 2024-10-05:
And yours CV are reversed regardless of using attenuators or attenuverters?

You could try removing the configuration files:

- connect your unit here https://www.rebeltech.org/patch-library/device/
- open the "resources" tab
- delete everything except oneiroi.cfg and restart
- optionally recalibrate once again

pssssssst, 2024-10-05:
Okay, i will try that this evening. Thanks for responding and also for releasing this module. It is so much fun.

pssssssst, 2024-10-05:
Okay, i couldn't wait :)
Your suggestion fixed the problem.
Thanks a lot. I'll keep testing further, but so far it seems fine.
This solution should probalby be posted as well over on modwiggler.

hirnlego, 2024-10-05:
Glad it worked! Yeah, when we'll release the new firmware we'll provide complete instructions. Thank you.

@rootapprox try flashing the new beta 2 and delete the configuration files as suggested above, this should fix your problem as well.

closed 2024-10-05 by hirnlego

rootapprox, 2024-10-05:
thanks a lot @hirnlego  i flashed the new beta and everything works fine , also i can save parameters now and stay persistant after the on/off power cycle.