Befaco/Oneiroi #3: Looper Issues

j3o3n3, 2024-08-21:
I noticed a couple of issues on Looper behaviour.

1) Trying to play a loop on reverse mode:
    - If start is in minimum value, loop is played correctly
    - If start is not in minimum value, loop only is played when lenght value is minimum or maximum.

2) Loop data is not played entirely on reverse playing when oneiroi is receiving sync
- For reproduce, rec a loop, play it reverse, and connect something to sync. Then place lenght control at maximum, and start control in one value over middle position.

j3o3n3, 2024-09-20:
Hi, only I wanted to clarify this bug is not only when oneiroi is receving sync. Also I described an issue when sync is not connected (point 1).  Thanks

hirnlego, 2024-09-29:
@j3o3n3  @Diegoterrores I think this should be fixed in the latest release (v1.1-beta), please check:
https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta