Befaco/Oneiroi #4: Looper Click

Diegoterrores, 2024-09-03:
The looper has an obvious click on the loop point.
Would be nice to implement some crossfade and/or cutting audio at zero crossing points to smooth this sharp transition.

WiretapStudios, 2024-09-09:
Also having this issue. The crossfade would help, or something like being able to change the end point like the start point knob to possibly cut the click off?

hirnlego, 2024-09-29:
@Diegoterrores I think this should be fixed in the latest release (v1.1-beta), please check:
https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta

WiretapStudios, 2024-10-02:
@hirnlego It solved the clicking for me, I've tried several dozen loops with no clicking. Can you outline what the expected result is if you either hold down the record or tap and then tap again - what happens when the loop is full? I made another post about it, but once it runs into the beginning of the loop, and SOS is all the way down, shouldn't it just overwrite? For me it gives this weird pitch stacking sound that is hard to work around.

hirnlego, 2024-10-03:
@WiretapStudios could it be that you're recording POST and varispeed is not quite at 1x? It could explain what's happening.
Anyway, please open a specific ticket for that issue, so we can close this one. And thank you for your feedback!

WiretapStudios, 2024-10-03:
@hirnlego I have one open already. It is not in post, I did check that each time.

hirnlego, 2024-10-03:
@WiretapStudios let's continue in the other one.

closed 2024-10-03 by hirnlego