Befaco/Oneiroi #5: Impossiblity of Mod Assign to secondary parameters

j3o3n3, 2024-09-03:
I try to assign mod to secondary parameters, Knob indicator marks the value is down, I trun knob to minimum but I can't pick up the value.

Maybe is only in my unit?

Here the image of the example. Trying to pick up the value with autopan knob at minimum while knob indicator marks value is below this value.
[image]

Diegoterrores, 2024-09-12:
It could be a calibration problem.
Is it only on this parameter?

j3o3n3, 2024-09-17:
Happens in almost all secondary parameters.
Only happens to me?

hirnlego, 2024-10-03:
@j3o3n3 you're talking about modulation but you're actually changing the CV mapping (green button). Are you trying to change the modulation or the CV mapping?

j3o3n3, 2024-10-06:
Yes @hirnlego as you say I'm trying to change the CV amount. (I realized after posting the thread, I was not able to correct the tittle). Anyway the result is same, I can't pick up the values as was described above.

hirnlego, 2024-10-17:
@j3o3n3 sorry for the delay... I cannot reproduce with the latest patch (beta 2), did you try with that?

hirnlego, 2024-11-10:
Closing for lack of activity, feel free to open another ticket if the issue is still present.

closed 2024-11-10 by hirnlego