Befaco/Oneiroi #6: Comb FIlter Noise

geraasmasjien, 2024-09-18:
When setting the filter to Comb Filter mode there is an audible noise occurring when the Frequency knob is at roughly the 8 o clock position. Below and above this position the comb sounds as expected but when at this position there are aperiodic clicks. I'm not sure if this behaviour is intentional but it doesn't seem to be consistent with the rest of the filter's response.

hirnlego, 2024-10-03:
I cannot reproduce continuous clicks if the cutoff is steady (no movement and no modulation). 
It's true that around 8 o'clock zapping sounds may occur when the cutoff changes, but that is because of the CF design when the frequency is low and I wouldn't define them as noise.
If you're experimenting this effect but continuously without changing the cutoff then it could be that either the knob or the CV are particularly sensitive. I would contact Befaco's support in this case.

geraasmasjien, 2024-10-03:
Thank you for the feedback. Upon receiving it I did some further testing. What I have found is that this behaviour is only present when the CV AMT is set to full (all the way clock wise) even when no CV input is present. When I turn the CV AMT on the CUTOFF to 0 it doesn't exhibit this behaviour and the comb behaves as expected.

I am happy to move this matter to support to figure out why it may be, I am guessing it could also be my power.

I did however find something interesting that I am pretty sure has to do with the software: 

Supplying the VCO V/Oct with 9-10V while the VCO is already transposed very high causes the SINE output to go out of the audible range and it starts aliasing back down into the audible range. This doesn't happen with the SS since it seems to be capped at a certain frequency. The sounds that this produces on the SINE output is actually very interesting however, after a while of doing this the SINE output stops producing any sound and doesn't respond to changes in CV or knob position, while the SS still seems to respond as expected and still makes sound.

hirnlego, 2024-10-05:
@geraasmasjien good catch! Fixed in beta 2, thanks.

closed 2024-10-05 by hirnlego