Befaco/Oneiroi #7: Looper picking up speed changes with new loop

WiretapStudios, 2024-09-19:
Awesome unit, I've logged many hours with it so far. However, I have ran into a few bugs that I really can't explain too well. 

One that I have had happen a few times is I'll clear the looper, with no modulation set anywhere yet, fresh from startup, no SOS turned up, etc. I'll clear the loop and then record a new one. The new loop will have a modulation type effect as if I was turning the varispeed or modulating it, but nothing is assigned yet, and I'm not touching any other buttons. I'm not sure how else to describe it, but maybe someone else has ran into this? 

A second looper related one is I see the sync light flash, but if that's tied to the looper, shouldn't it start at the time you hold it down, and then flash again after 5 seconds to indicate the sync time of the loop from the start point? If not, that would be handy to have an indicator on when the full 5 seconds was up. 

A third thing I have ran into a few times is the looper recording a "post" oscillator sound even though the looper is set to "pre" and the sine and ss/wt sliders are fully down. It could be from the fx, but I'm not sure how that would work if it's set to "pre" per the manuals description. It's like something has bled into the looper. 

I may be completely misinterpreting these as bugs instead of features, so no worries if that's the case. I did read the manual through a few times.

hirnlego, 2024-10-03:
> One that I have had happen a few times is I'll clear the looper, with no modulation set anywhere yet, fresh from startup, no SOS turned up, etc. I'll clear the loop and then record a new one. The [...]

I cannot reproduce. The effect you're describing usually depends on the things you already checked (varispeed modulation, SOS when speed is <> 1, or when recording POST with speed <> 1).  
Maybe the module's configuration is incorrect, please try this: 
- connect your unit here https://www.rebeltech.org/patch-library/device/
- open the "resources" tab 
- delete everything except oneiroi.cfg and restart
- if nothing changed, try to recalibrate the unit

Could you record a video of this issue?

> A second looper related one is I see the sync light flash, but if that's tied to the looper, shouldn't it start at the time you hold it down, and then flash again after 5 seconds to indicate the [...]

When there's no external sync, the led flashes each time the buffer restarts, it's frequency is fixed and unrelated to the looper's parameters.

> A third thing I have ran into a few times is the looper recording a "post" oscillator sound even though the looper is set to "pre" and the sine and ss/wt sliders are fully down. It could be from the [...]

Cannot reproduce this either. PRE and POST are two different sources entirely, there cannot be bleeding from one to the other. Could you provide a video for this too?

WiretapStudios, 2024-10-09:
@hirnlego would you rather links here or have them emailed to you?

hirnlego, 2024-10-09:
@WiretapStudios if you can post/link them here, thanks!

WiretapStudios, 2024-10-26:
OK, so I've been working with it and I think resetting the configuration files fixed a thing or two that must have been in there when I received it (I purchased used). 

I did test and the length knob did shorten the sample on both ends, not just adjusting the end point of the sample. If I recorded a spoken phrase and the start was fully left, and length was fully right, once you start moving the right knob towards the middle, the start position is also moving, shortening the phrase. Does it sound like I should calibrate again? I can also get a video of that one if needed.

The two things that I currently can't eliminate are bleed from the levels of the looper/sine/ss/wt when all the faders are down (recalibrate again?):

https://drive.google.com/file/d/1W_cE_6_TjnFPI6ZTUDkQuDMMIcAdm8-r

And here is an example the sampling where it stacks on itself:

https://drive.google.com/file/d/1WeAHpXvziwmpYN7Twdo_OHnb-DtNg5ik

WiretapStudios, 2024-11-05:
Scratch the looper stacking, I see my mistake now, I was creating an audio loop through my audio interface / sample source how I had it set up and it was doing sound on sound that way. 

I also somehow eliminated the persistent sound bleed after some restarts, not sure how. I'm going to close this up and recalibrate it when I get a chance just to make sure.

closed 2024-11-05 by WiretapStudios