Befaco/Oneiroi #8: Loop length question/suggestion

WiretapStudios, 2024-10-02:
Loop length - I'm not sure if I'm thinking of this in the wrong way, but why does the length cut off from both ends if the start point is adjustable? For other synths I have that trim samples, the length would adjust the endpoint only since you can already move the start point in the loop to shorten it. It makes it hard to trim off silence at the end of a loop while keeping the start point intact.

I'd love to see an option in the startup settings to change the length knob to an "end point" adjustment. 

Overall this thing is a monster, I've been using it nonstop since I got it and you can get some incredible soundscapes out of just this module, kudos.

hirnlego, 2024-10-05:
@WiretapStudios length should only affect the end point, not sure why in your case it also affects the start point.

WiretapStudios, 2024-10-08:
@hirnlego , great, thanks for confirming, I may have been perceiving it differently, I tried it again a few times and confirmed now it's only moving the end point, not sure what I was hearing the other times.

closed 2024-10-08 by WiretapStudios