Befaco/Oneiroi #9: Echo ( delay ) Density knob mode as Repitch type instead of Jump of the rate.

rootapprox, 2024-10-07:
Hello i would like to suggest to have the Density knob in the Echo goes smoothly in the rate without that aliasing sound heard when you change the Rate , i think maybe the Delay transition mode should be in Repitch mode ?
PS : this could be leading to other thing in the Delay ( echo ) and its not related at all to the Delay transition mode, but i think you got my idea of having the Density Knob moves smoothly within the rate values.

geraasmasjien, 2024-11-04:
Agreed that it would be great if the echo didn't produce the clicky/aliasing sound when Density in changed.

hirnlego, 2024-11-10:
@rootapprox try the new beta 5, it should be smooth now, and without the pitch shifting effect. Let me know!

rootapprox, 2024-11-10:
@hirnlego sounds good as i dont hear the Click/Aliasing sound when changin the knob , but i would love to hear the pitch shifting effect ,( that was the main reason to open this issue) Like Mimeophone Style , i gave an analogy from Ableton Live delay where the delay desired is Repitch Mode unstead of Jump Type..

Great work so far on this!

rootapprox, 2024-11-10:
some reference here from Clouds Parasite firmware ( Looping delay and Pitch-shifter mode ) maybe you can inspire by its code.

hirnlego, 2024-11-10:
@rootapprox I get what you say and actually I feel that I removed a cool characteristic of the echo, now glitchless but soulless, so I reworked it once again and now I think I found the right balance. Please try downloading the beta 5 once again, I swapped the file.

hirnlego, 2024-11-10:
> some reference here from Clouds Parasite firmware ( Looping delay and Pitch-shifter mode ) maybe you can inspire by its code. https://mqtthiqs.github.io/parasites/clouds.html#pitch

I'll have a look, even if I won't touch anything else of the looper for the time being. Thanks for the reference, though!

rootapprox, 2024-11-10:
sorry to bother but when i have the Echo controlled by the SYNC in clock im sending , there is a audible click happens when the clock hit from the delayi confirm this happens only when u send a Clock to Sync IN

hirnlego, 2024-11-11:
Yes, I confirm this. Working on a solution.

hirnlego, 2024-11-11:
@rootapprox should be fixed in the new beta 6. Please check it out and report any issue in the relative announcement ticket, thank you. I'm closing this one as fixed.

closed 2024-11-11 by hirnlego