Befaco/Oneiroi #11: Looper click returns at slower speed (beta 1.1)

SetProxy, 2024-10-18:
Hello!

DIY, calibrated, updated v1.1-beta. 
At 1x speed, there is no click and the loop crossfade seems to work wonderfully, but at 0.5x or lower speeds the click returns very prominently and wont change for the better even when starting position or length is changed. When returning to 1x speed, crossfade works nicely again. Not sure if bug or works as intended or I am doing something wrong. It would be wonderful if there is smooth crossfade even at lower speeds.
I appreciate the work, wonderful module!

hirnlego, 2024-10-20:
@SetProxy I fixed a couple of things related to the crossfade, could you check with the latest beta 3?

https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta.3

SetProxy, 2024-10-20:
I updated to beta3 and re-calibrated just in case.
It is definitely an improvement, there is significantly less click and I can now reduce it even further with adjusting the loop start position and length. 

While testing it out, I noticed a weird behavior of the loop start position knob. 
I recorded a loop of an oscillator going low to high pitch. I turned the loop length knob to 12:00 clock and slowly started to scan the loop from the beginning. Around 12:00-13:00 the start knob lost the correct position in the loop, did a small jitter further forward and around 14:00 returned audio playback to the beginning of the loop. When turning the knob from 14:00 to 17:00, it continued to scan the loop from the beginning.

hirnlego, 2024-10-22:
Please, open another ticket for that, this one is closing now.

closed 2024-10-22 by hirnlego