Befaco/Oneiroi #12: Beta 3 for v1.1

hirnlego, 2024-10-20:
Beta 3 is out!

https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta.3

If coming from v1.0, a configuration erase is required:

- connect your unit here https://www.rebeltech.org/patch-library/device/
- open the "resources" tab
- delete everything _except_ oneiroi.cfg and restart
- optionally recalibrate the unit

@Diegoterrores I fixed a couple of important things in the latest beta, could you pass it to everybody involved in beta-testing that aren't active here, please?

markcrosbie, 2024-10-22:
I'm on firmware v1.0 and I'm trying to upgrade to this firmware. But the update never seems to finish. I deleted all of the config files except oneiroi.cfg, and then flashed the firmware. It gets as far as block 100 and then stalls. Never seems to complete.

Mac OS Sonoma 14.7 M1, Chrome Version 129.0.6668.101 (Official Build) (arm64

Firmware version: Oneiroi v1.0
Preset 0: Oneiroi [99268][0x0]
Preset 1: Oneiroi [99268][0x7367c65f]
Reading file patch.syx
Sending data 118210 bytes
Program Message: Loading 30
Program Message: Loading 33
Program Message: Loading 79
Program Message: Loading 100

Diegoterrores, 2024-10-24:
> could you pass it to everybody involved in beta-testing that aren't active here, please?

Yes! I'm on that

hemmer, 2024-10-25:
One thing: unison can be negative but the clamping of sine frequencies is always positive so the lower half of the knob doesn't produce the second tone. 

https://github.com/Befaco/Oneiroi/blob/main/StereoSineOscillator.h#L91

hemmer, 2024-10-27:
Actually I'm not too sure what expected behaviour of sine is? At 12 oclock (with notch) I get a beat pattern (because one sin is ~0 * f which is clamped to 16 Hz (kOscFreqMin)), below no change and above a rising sine until the pitch matches. I don't get an octave between at max knob value either, just the same frequency. 

Manual says "the frequency of one of the two sine’s channels [...] are lowered one octave down at 0% and raised one octave at 100% (default is 50%, unison)"

hirnlego, 2024-11-01:
@markcrosbie did you finally manage to flash the firmware or it's still giving you troubles?

markcrosbie, 2024-11-01:
@hirnlego it worked! I was able to update to Beta4, calibrated the module and now the V/oct input correctly tracks pitch against CV

hirnlego, 2024-11-01:
> @hirnlego it worked! I was able to update to Beta4, calibrated the module and now the V/oct input correctly tracks pitch against CV

Great!

closed 2024-11-04 by hirnlego