Befaco/Oneiroi #13: Loop start position knob issue

SetProxy, 2024-10-23:
I noticed a weird behavior of the loop start position knob.
I recorded a loop of an oscillator going low to high pitch. I turned the loop length knob to 12:00 clock and slowly started to scan the loop from the beginning. Around 12:00-13:00 the start knob lost the correct position in the loop, did a small jitter further forward and around 14:00 returned audio playback to the beginning of the loop. When turning the knob from 14:00 to 17:00, it continued to scan the loop from the beginning.

hirnlego, 2024-11-01:
@SetProxy it's hard to reproduce your specific issue, but I tried with a vocal sample (that makes easier to notice the loop's start and end points) and I'm not finding anything wrong with the start knob. Could you maybe try recording a vocal sample, like a speech, and check again?

hirnlego, 2024-11-10:
@WiretapStudios may be an issue with the start knob resolution. I raised it a bit in the latest beta 5, could you try and see if solves the issue?

WiretapStudios, 2024-11-17:
So far, no updates have fixed this one.

WiretapStudios, 2024-11-17:
I did bring it up in other threads, but since this one is still open, I'm having the same issue as the original poster. 

I have recalibrated 3 times. The start knob is not changing the position of the start. The Length knob begins to cut the length, but also cuts the start point turning left. At certain points it cuts the middle and not the start or end points.

hirnlego, 2024-11-17:
@WiretapStudios do you mean that START has no effect whatsoever on the loop?

WiretapStudios, 2024-11-17:
@hirnlego Yes, at almost any position while turning you can hear it scrub but it's still at the same start point. I could also provide a video of some talking only audio to show where the end points are if that helps to see what I mean.

hirnlego, 2024-11-17:
@WiretapStudios that's very weird. Please, send a video!

WiretapStudios, 2024-11-20:
OK, I will work on getting one to you shortly.

WiretapStudios, 2024-11-23:
@hirnlego OK, here is an example, the sample is a person saying A B C D in order. The Start knob  position doesn't move through the sample almost at all. The Length creates odd positions, this time saying A and D but  nothing in the middle. 

I have calibrated the knobs several times, the others all seem to work. In the calibration process I move the Start and Length the same way as the other knobs, I'm assuming that is not a mistake. No idea where to go from here.

https://drive.google.com/file/d/1LOdDaGHbt6A5OZg4Kx_BTlfv9_lv2HAn/view?usp=sharing

WiretapStudios, 2024-11-25:
@SetProxy does my file sound somewhat similar to what was happening with you?

SetProxy, 2024-11-26:
Hey!

Sorry for being inactive, I sent my DIY Oneiroi to Befaco for Happy Ending repair and calibration. Currently waiting for repairs to finish.

I took a look at the video, this looks more extreme than my situation. 
At full loop length, turning the start knob did change the loop start point relatively accurately and fast, until around 12:00-13:00 clock. Then the loop start point jumped around for a moment and went back to the beginning of buffer. Turning the knob further from 13:00 made the looper scan the first half of the buffer.
Changing the loop length did change the length as expected.

I will resume testing when I get my Oneiroi back

WiretapStudios, 2024-11-29:
@hirnlego please check out my file example above and advise if there is anything I can do for that as it's a pretty big part of the unit that isn't working.

hirnlego, 2024-11-29:
@WiretapStudios well, I think that actually it seems to work as intended. 

The start control only makes a difference when length is not 100%, or when the looper is externally triggered, otherwise the loop always sounds the same regardless of where it starts. 

You can try two things to test if START works:

- try reducing the length and then changing the start position, it should change the section of the buffer that is being looped
- apply a SYNC IN. Changing START should change the point where the loop restarts with each sync.

As for the fact that in your case reducing the length leads ultimately to silence, depends on the fact that probably your recording begins with silence, and when you reduce the length to just the first part, you get silence.

WiretapStudios, 2024-12-04:
@hirnlego I could be completely dense about this, but I don't understand the advantage of that vs start and end points like all the other loopers, samplers, granular, etc. hardware that I have. Why would one want a less precise way of choosing what section is being looped? 

When I previously asked, I understood the answer to be that the length doesn't change the start and end points at the same time. But, if it's not moving just the end point, then wouldn't that mean it's shrinking it from both ends? Also, would a workaround for not having a real start and end point be to move the length to approx. 99% and then move the start knob from there? 

Apologies if this seems like splitting hairs but I'm just trying to parse this out so I can get the type of loops I want when initially hitting the sample button.

hirnlego, 2024-12-04:
> Apologies if this seems like splitting hairs but I'm just trying to parse this out so I can get the type of loops I want when initially hitting the sample button.

No apologies, but this should really be straightforward, I'm not sure if we're using different names for the same things or the same name for different things.

START sets the start point of the loop, LENGTH sets the length of the loop. The end point of the loop is derived from the start point and the length.

Hopefully an illustration will clear things up:

[image]

There are 4 scenarios, all with a buffer filed with the recording of a voice that spells "A... B... C... D...". For simplicity the letters are equally distributed in the buffer and each one starts exactly at every quarter of the buffer.

1. With START = 0 and LENGTH = 100% the buffer is reproduced fully, the audio loops as "A... B... C... D...".
2. With START = 25% and LENGTH = 100% the buffer is still reproduced fully but starts at "B", the audio loops as "B... C... D... A...". The effect is the same as the above scenario, you simply can't tell where the loop starts unless you trigger it externally. 
3. With START = 25% and LENGTH = 50% the audio loops as "B... C...". Now you can definitely tell that the loops starts at "B".
4. With START = 50% and LENGTH = 75% the audio loops as "C... D... A...".

HTH

znakebyte, 2024-12-04:
Hi,
Loving the Oneiroi and I've been following this discussion with a keen eye.

That illustration and explanation should be in the manual, very helpful 👌🏻

Kind regards
Kristoffer E. M. Amundsen

WhatsApp.Image.2024-12-04.at.19.36.43.jpeg (view on web)<https://github.com/user-attachments/assets/2f094d16-77d1-4aea-b20a-2ef033834209>

WiretapStudios, 2024-12-05:
@hirnlego The illustration helps, and I agree with znakebyte that something like that should be illustrated in the manual. 

I understand the trigger in the second (B C D A) starting at the B point. But having the beginning of a loop play starting the second loop seems like such a small use case vs. having control of the full loop length and start/end points. It's more experimental but less useful, start point for a trigger should be a secondary function to actual loop start/end points IMO since you're not always going to be using it that way (triggered). 

That aside, I have tried manipulating loops like you are describing in your illustration with the same knob percentages, and it's still not functioning like that. I'm using full samples with people talking and no silence. Moving the knobs results in random parts being looped that don't follow the illustration's logic. I think that might be the biggest problem that's causing the above to not "click" with me because I'm not able to trim the loops as shown to get the portion I want. I'm still not sure if it's a hardware or "me" issue.

Since the 5-ish second timing is not exact, you can manually time it by letting go of the record after the part where you want it to loop. Perfect. Now to trim the start point to remove a part that's been clipped. How do you remove the first half a second? Again, sorry if it seems like I'm splitting hairs, but this is the first sample/looper that I have owned where you can't manipulate the start point (without a trigger). 

I love the module though, I've been using it for a month now almost every day. That's why I'm finding this part a little baffling as it doesn't seem to be that complex of a concept, even if it's not the way I would prefer it to work. 

Sidenote: I appreciate all the hard work with the updates, any other issues I initially had are all now gone.

hirnlego, 2024-12-05:
@WiretapStudios the START and LENGTH parameters shouldn't produce random results, I get the predictable and consistent result of the illustration when turning the knobs. I'm not aware of other users having the same issues, so I'm beginning to think that something's not right with your unit...

WiretapStudios, 2024-12-14:
@hirnlego what would you suggest? I contacted the person I bought it from on Reverb and he bought it premade. He mentioned a few issues he had but those were bugs that I believe have been worked out already in the updates.

WiretapStudios, 2024-12-14:
@hirnlego OK I will do that - sorry, I wasn't sure how involved with the company itself you were.

hirnlego, 2024-12-14:
@WiretapStudios no worries!

Diegoterrores, 2025-03-11:
This has been added to a new draft of the manual and will be published soon. 
In the meantime, could you @hirnlego check the manual on Drive where we added your explanation to confirm that it is okay before we publish the new version?

hirnlego, 2025-03-12:
@Diegoterrores left a comment in the doc.

Diegoterrores, 2025-03-13:
This information is now in the manual, so I will close the issue. 

You can check the [manual revision here](https://befaco.org/docs/Oneiroi/Oneiroi_User_Manual.pdf
)

Any further manual suggestions can be addressed to:

support@befaco.org.

closed 2025-03-13 by Diegoterrores