Befaco/Oneiroi #17: Beta 4 for v1.1

hirnlego, 2024-11-01:
Here:

https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta.4

@Diegoterrores things that have been modified and need to be tested:

- volumes of the sources (I've raised a bit the input level and tweaked the oscillators that were a little irregular)
- I've lowered the filter of the V/OSC input but I'm not sure if the change has actually translated to a different response
- unison control should work correctly now

Also, from the previous beta 3:

- looper's loop point should produce noticeable less click or none at all
- wavetable should sound less harsh and offset control should be smooth (the CV also)

If everything works as it should, I think we may have a release candidate.

Diegoterrores, 2024-11-01:
Passing the announcement to beta testes!

hemmer, 2024-11-02:
Unison is sounding great, thanks!

Re input level I feel like looper resampling is maybe clipping a bit now? `resample_` buffer now gets x 2 gain [here](https://github.com/Befaco/Oneiroi/blob/main/Oneiroi.h#L218) exiting the loop, then x 1.6 [here](https://github.com/Befaco/Oneiroi/blob/main/Looper.h#L395)? The limiter stops it going crazy but it sounds a bit crunchy to me. Am I missing something about these changes (https://github.com/Befaco/Oneiroi/commit/7c262de0a70bd580047f8a1b3c919e2449570a77#diff-8b258dc52ea186fbfed381dac58d02a312a7251c6e3f112053a34ed968f1be80R218)?

cdromain, 2024-11-10:
Hi Roberto @hirnlego 

Congrats on this ambitious and very special project 🙌🏼

Got my factory Oneiroi yesterday (I imagine firmware was 1.0), tried to install beta 4 and calibrated the module, but would need clarification on the firmware update procedure please :

1. I read on the issue #12 that a configuration erase was needed ? Could you please confirm if it's still the case with beta 4 and what the order is ? (erase then install beta firmware ?) This step is missing from the firmware update procedure.
2. What is the expected feedback in the OpenWareLaboratory web interface when the firmware update is finished ? Is `Program Message: Loading 100` the final message confirming that firmware installation is done & ok ?
3. Any way to check the firmware version installed, either on the module itself or the OpenWareLaboratory interface ?

I think making these clear in the documentation would be helpful.

I'd be happy to help and contribute to the module development and improvement so don't hesitate to reach out for questions, (beta) tests or anything. I'd happily join the beta testers crew if possible (👋🏼 @Diegoterrores)

Thanks in advance for your feedback and keep up the great work 🚀 
Cheers,

Romain

hirnlego, 2024-11-10:
> Unison is sounding great, thanks!
> 
> Re input level I feel like looper resampling is maybe clipping a bit now? `resample_` buffer now gets x 2 gain [here](https://github.com/Befaco/Oneiroi/blob/main/Oneiroi.h#L218) exiting the loop, then x 1.6 [here](https://github.com/Befaco/Oneiroi/blob/main/Looper.h#L395)? The limiter stops it going crazy but it sounds a bit crunchy to me. Am I missing something about these changes ([7c262de#diff-8b258dc52ea186fbfed381dac58d02a312a7251c6e3f112053a34ed968f1be80R218](https://github.com/Befaco/Oneiroi/commit/7c262de0a70bd580047f8a1b3c919e2449570a77#diff-8b258dc52ea186fbfed381dac58d02a312a7251c6e3f112053a34ed968f1be80R218))?

You're right, I tweaked them again in beta 5. Would you mind check that one?

hirnlego, 2024-11-10:
> CV on Wavetable Offset works as expected. Wonderful! Really nice to be able to modulate this parameter smoothly.
> 
> Curious about the fluctuation in volume when scanning through the wavetable and if there is a way to keep the volume stable across the travel of the knob or CV input?

I've tried compensating this effect with gain and saturation, I think it's better now but in the end it all depends on the recorded material. Check the latest beta 5.

hirnlego, 2024-11-10:
>     1. I read on the issue [Beta 3 for v1.1 #12](https://github.com/Befaco/Oneiroi/issues/12) that a configuration erase was needed ? Could you please confirm if it's still the case with beta 4 and what the order is ? (erase then install beta firmware ?) This step is missing from the firmware update procedure.

Erasing is needed when coming from v1.0, if you did it already then there's no need to do it again when upgrading to another beta and won't be needed when finally upgrading to v1.1.
I don't think there's an order to follow, you can upgrade and erase or the other way around.

>     2. What is the expected feedback in the OpenWareLaboratory web interface when the firmware update is finished ? Is `Program Message: Loading 100` the final message confirming that firmware installation is done & ok ?

The output should be something like this:

`Reading file patch.syx
Sending data 119916 bytes
Program Message: Loading 35
Program Message: Loading 43
Program Message: Loading 80
Program Message: Loading 88
sent patch 1
Program Message: Loading 100
Preset 1: Oneiroi [100700][0xfdbd30ab]`

>     3. Any way to check the firmware version installed, either on the module itself or the OpenWareLaboratory interface ?

Unfortunately the only way to update the version number in the web interface would be to change the actual firmware and not only the patch, and would require flashing two files instead of just one (feasible but maybe an hassle for the user).
It might be easier to come up with some LEDs code on module's startup, I'll think about it.

cdromain, 2024-11-10:
Thanks for your quick reply @hirnlego !

> Erasing is needed when coming from v1.0, if you did it already then there's no need to do it again when upgrading to another beta and won't be needed when finally upgrading to v1.1. I don't think [...]

Had not actually erased the config files yesterday (wasn't aware of that required step).

Installed the new beta 5 tonight and got the expected output in the [OpenWare Lab](https://pingdynasty.github.io/OwlWebControl/extended.html) interface.

Keeping Oneiroi connected with USB, I then switched to the [other website](https://www.rebeltech.org/patch-library/device/) and there was actually nothing to delete - `oneiroi.cfg` was the only file listed, so I did not have to erase any file. Maybe my module (latest batch) already came with a post 1.0 firmware ?

I then rebooted the module and ran the calibration procedure again.
I guess it's all good and I'm running beta 5 👍🏼
Testing it right now, will post some feedback around here. Cheers !

closed 2024-11-10 by hirnlego