Befaco/Oneiroi #18: Beta 5 for v1.1

hirnlego, 2024-11-10:
Here:

https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta.5

Apart from a few small tweaks here and there, I've added fade in and out when the recording starts and stop, so to further avoid clicks. I've also reworked how echo's density control works and now it's completely free of aliasing (losing the pitch shifting effects along the way).
The patch now uses a bit less CPU that could be devoted to something else in case it's needed.

I promise I won't touch anything else anymore for this release, unless any other bug emerges, of course.

Let me know!

hirnlego, 2024-11-10:
Well, I gave a step back about the pitch shifting and managed to restore it without glitches or artifacts, I like how it sound now. I replaced the release's file, it should be downloaded again.

geraasmasjien, 2024-11-10:
Wavetable sounds great! Thanks for the wonderful work.

Also glad you found a balance on the echo density that still has a character that you are happy with. I think it sounds good and quite unique.

cdromain, 2024-11-10:
@hirnlego Testing beta 5 tonight and loving it, kudos again - and I'm glad you finally kept the pitch-shifting when changing echo density !

The looper is mostly click-free, even when changing speed (crossfade at start/end points does seem to work quite well) and start - but I'm getting loud clicks as soon as I shorten the loop length. Thought it had been fixed in previous betas ? (just got my Oneiroi yesterday)

Dreaming about a click-free looping experience 🙌🏼

Please advise, I'm ready to run more tests. Thanks !

WiretapStudios, 2024-11-11:
I'm getting clicks when looping. I've tried both holding the button down and turning it on and off. I also tried reducing the input level and also used an ambient track so there was no percussive sounds, and there is an audible pop/click in the loops. I tried about 20 loops and only got one without an audible click. The loops themselves sound great until the click.

Side thought, is there a way to implement that when the record loop buffer is full, the button light/record turns off automatically?

hirnlego, 2024-11-11:
> I'm getting clicks when looping. I've tried both holding the button down and turning it on and off. I also tried reducing the input level and also used an ambient track so there was no percussive [...]

Strange. Were you using external sync? Was echo or any other effect active? 

> Side thought, is there a way to implement that when the record loop buffer is full, the button light/record turns off automatically?

That's quite a specific need, I for one wouldn't want the record to stop. Feel free to open a feature request, though, we never know.

WiretapStudios, 2024-11-11:
For mine, it's the full length loops with start and end points fully open. Shorter ones I can understand if there are clicks. No fx or anything else enabled, just audio in to a dry new loop after restarting to test.

cdromain, 2024-11-11:
What I described yesterday were clicks happening when the loop length was getting shortened, so not with very small loops only. Material recorded was POST (i.e. internal oscillators through the FX). Oneiroi was unsynced and there was almost no modulation (only some internal on Filter Freq).

Tested more this morning and could reproduce the issue, even if it does not seem to be 100% systematic. The issue might be very sensitive to the material in the looper buffer.

My test case today :
- Recording POST
- Sine and SS
- All four FX > 50% wet
-> Clicks when moving length CCW, especially shortening after 3PM (does not seem to be dependent on buffer speed)

Will keep testing (loving it :)

hirnlego, 2024-11-11:
@cdromain @WiretapStudios please, try the new beta 6, hopefully the looper will be click-free now.

I'm closing this issue, please report any problem in the new announcement, thank you.

closed 2024-11-11 by hirnlego