Befaco/Oneiroi #21: Feature request - randomize but do not change pitch

cdromain, 2024-11-11:
Loving the onboard randomizer ! 🎲

But I really wish we could randomize ALL without changing the oscillators pitch.
The usecase is simple - if Oneiroi is playing with other voices in a common pitch/scale, I don't want its pitch to randomly change. But I'd still like to have all the other sections being randomized.
Right now I can't, as it means choosing as target LOOPER or FX (i.e. can't have both without randomizing pitch).

Given that randomization targets are hardcoded on the front panel, I guess the only option would be to add an option to disable pitch change in the randomization (in the startup settings ?)
I'd be fine with the oscillators pitch being excluded from randomization, but I understand some might like it when using Oneiroi on its own.

(Pitch change could be made more musical with a choice of scales but that might be hard to implement in the current UI.)

Happy to brainstorm more and test !

Thanks 🙌🏼

hirnlego, 2024-11-12:
I was thinking about an alternative way of handling startup options, and in the case I found one this could be one of those, I agree it could be useful.

Diegoterrores, 2025-03-11:
This would be possible by implementing the feature request as issue #39 (sysex access for options)