Befaco/Oneiroi #22: Beta 6 for v1.1

hirnlego, 2024-11-11:
Here: 

https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta.6

Both echo's clicking when sync'd and looper's clicking when looping should be fixed.
I also improved the fade when the looper is retriggered, to avoid nasty clicks.

rootapprox, 2024-11-11:
Great ! trying it now!

rootapprox, 2024-11-11:
I noticed some weird pitch shifting occurs with the SYNC IN used ,( tried without synced clock and it works fine )
here an example of a little loop with just Input UP and Echo Fully wet. you will hear some little transitions of pitch from time to another in the loop 
[8-Audio 0001 [2024-11-11 172426].wav.zip](https://github.com/user-attachments/files/17709095/8-Audio.0001.2024-11-11.172426.wav.zip)

hirnlego, 2024-11-12:
@Diegoterrores I've added a few details in the release's notes that should also be added to the manual, specifically the fact that the version number in the web page doesn't change (it shows the firmware's version, not the patch's). We should think of a way to display the patch's version, maybe using the LEDs on startup.

rootapprox, 2024-11-12:
> @rootapprox were you changing DENSITY? Changing DENSITY causes a pitch shift, regardless of the way it is changed (knob, CV, modulation), and when the echo is sync'd the effect is that a short pitch shift happens when stepping from a time division to the other. The solution to this would be removing pitch shifting from DENSITY when echo is sync'd, I thought about it but I like this effect.

Nope i was not changin the DEENSITY , it is fixed when this happens ...

hirnlego, 2024-11-12:
> The looper is working great! Probably the best I've heard it so far, I tried several dozen and made some really nice stuff. Adjusting the start and end does introduce clicks at many spots, but if that's the nature of the beast then I'm OK with at least the full loop being clean.

By "clicks" you mean actual waveform discontinuities? And do those clicks continue when you stop modifying START and/or LENGTH? Is the change of the parameters the cause or is it the loop point?

hirnlego, 2024-11-12:
> > @rootapprox were you changing DENSITY? Changing DENSITY causes a pitch shift, regardless of the way it is changed (knob, CV, modulation), and when the echo is sync'd the effect is that a short pitch shift happens when stepping from a time division to the other. The solution to this would be removing pitch shifting from DENSITY when echo is sync'd, I thought about it but I like this effect.
> 
> Nope i was not changin the DEENSITY ( nor knob, CV or Mod) , it is fixed when this happens ...

I can't reproduce it on my unit if DENSITY is not touched in any way. @WiretapStudios are you experiencing this issue as well?

rootapprox, 2024-11-12:
> I can't reproduce it on my unit if DENSITY is not touched in any way. @WiretapStudios are you experiencing this issue as well?

Are you sure you are having a clock sent to the SYNC IN ? it happens when only there is a clock in the SYNC IN , i sent you a wav file explaining that sound ( it sounds like a laser gun fx )

hirnlego, 2024-11-12:
@rootapprox yes, I'm syncing the module.

After checking the videos you sent me it's evident that there's an issue there, the problem is replicating it on my side. 
In your case it sounds like as if you didn't update to beta 6 o you were modulating DENSITY somehow, but as you confirmed to me that you updated correctly and cleared every modulation, I'm a bit lost.
Let's see if any other beta tester has the same issue, in meantime I try to investigate a bit more.

rootapprox, 2024-11-12:
@hirnlego Yes i confirm that i updated to the latest plus did the calibration just in case , removed all the files from the resources just the cfg one ... i suspect this happened when you tweaked the Sync IN click of the density , for now im using it without SYNC IN clock to avoid the laser repitch.

hirnlego, 2024-11-12:
@rootapprox what you're experiencing is just the click of the previous beta but with the pitch shift effect. 
The click is (was) the delay time that jumps around due to instability in the clock reading, thing that should have been mitigated by applying averaging and some logic to the operation. It's possible that in case of slow clocks this fix is not enough, I have to check again. Thanks for the report!

rootapprox, 2024-11-12:
Yes as i read in the manual _ It is advisable to use a clock with a trigger per bar or less, such as a trigger every four bars or so, as the same clock is used for these three different tasks, including resetting the looper. _
so i programmed my beatstep pro to send a slow clock ( trigger each 4 bars ) and the for now if i want to get rid of that laser sound of the delay i have to choose fast clocks , like one each bar , and its gone the problem, hope everything i described helps in a way ! Great work !

cdromain, 2024-11-12:
> @Diegoterrores I've added a few details in the release's notes that should also be added to the manual, specifically the fact that the version number in the web page doesn't change (it shows the [...]

Thanks, much appreciated Roberto !

hirnlego, 2024-11-12:
> Yes as i read in the manual _ It is advisable to use a clock with a trigger per bar or less, such as a trigger every four bars or so, as the same clock is used for these three different tasks, [...]

As long as the clock received by Oneiroi is >= 30 BPM you should be fine and not experience any zapping effect. 
I tried with a 30 BPM clock and had no problem, with a slower clock things begin to fall apart, including the zapping effect. You're most probably using a clock too slow.

Anyway, I'm still on the fence about keeping the pitch shifting when externally sync'd, I'll see.

cdromain, 2024-11-12:
Hey guys, reporting after installing and testing beta 6 tonight.

The great news : the looper clicks situation is much much better and it's now an almost click free experience, great job @hirnlego 🙌🏼

Unfortunately I still got some clicks during my tests (albeit a lot less than with previous betas).
> I am not referring to artifacts when changing parameters, but repeated "looping clicks", I imagine around the loop point.

More details :
- The test case is still the same : unsynced, no modulation, recording (POST) internal oscillators (sine + SS) with or without FX (FX around 50% wet).
- Playing with **START** does not introduce clicks and the artifacts when changing it are actually pretty cool ! (nicer, smoother than previous betas it seems)
- Changing the loop **LENGTH** by itself (i.e. with **VARISPEED** stable) does not seem to introduce looping clicks (only single clicks when actually changing the value - would be nice to make them disappear though, as they are pretty loud).
- Looping clicks can appear when playing with both **VARISPEED** and **LENGTH** (LENGTH between fully CW and noon), like they are the result of interactions between these 2 parameters being changed. I struggle to find a reproducible pattern. It seems they tend to appear after heavy manual manipulation of these 2 parameters.

Hope it's helpful and we can make them go away for good !
Love Oneiroi and its looper so much, so playable I can't stop 🔥

hirnlego, 2024-11-13:
@cdromain if I understand correctly you are recording POST with **also** the looper present in the mix, so what you're actually recording is like having sound on sound on but with also the two oscillators, right?
If this is the case, I think that there is too much going on there to expect a recording without any type of audio discontinuity. Do you have the same issues when recording PRE?

cdromain, 2024-11-13:
> @cdromain if I understand correctly you are recording POST with **also** the looper present in the mix, so what you're actually recording is like having sound on sound on but with also the two [...]

Hey, no I was only recording POST with oscillators into FX (looper not in the mix).
Will test with PRE tonight and report back 👍🏼

hirnlego, 2024-11-13:
@rootapprox @cdromain please check the latest beta 7 and report on your issues here https://github.com/Befaco/Oneiroi/issues/24, thanks! 

I've made a couple of changes and hopefully at least the zapping effect should have been eliminated, this time I had to remove the pitch shifting effect when the module is sync'd, but it's better for this scenario.

I've also smoothed some more the looper's controls. 

Closing this one!

closed 2024-11-13 by hirnlego

cdromain, 2024-11-13:
I'm on it man 🔥

rootapprox, 2024-11-14:
> @rootapprox @cdromain please check the latest beta 7 and report on your issues here #24, thanks!
>
> I've made a couple of changes and hopefully at least the zapping effect should have been eliminated, [...]

Perfect !! i guess thats why in some nice delays around the pitch shifting its not there when its Synced with a clock, great job !