Befaco/Oneiroi #24: Beta 7 for v1.1

hirnlego, 2024-11-13:
Here:

https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta.7

- changed the way DENSITY is handled when the module is externally sync'd, removing the pitch shifting effect that in this scenario can be overwhelming
- further smoothed looper's controls

cdromain, 2024-11-13:
Quick remark : https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta.7 notes should point to #24 https://github.com/Befaco/Oneiroi/issues/24 (and not #22)

cdromain, 2024-11-13:
Reporting back after testing the fresh new beta 7 - to be honest did not seem much different from beta 6.

> Please refer to my comment yesterday for details on the test scenario : https://github.com/Befaco/Oneiroi/issues/22#issuecomment-2471695244

Had a few loud non-repeating clicks as I was changing the values (same as beta 6).
And like with beta 6, I ended up finding spots on my 2 axes of looper manipulation (LENGTH & VARISPEED) where the repeating looping clicks appear - guess that makes me a a bit of a clicks digger 😅

And I tested with PRE (recording the output of an external wavetable VCO) as you suggested with the same results.

An interesting find (cannot say if it's specific to beta 7 vs 6, I'd bet it's not) - 3PM on both params seems like a gold mine for those damn clicks !

rootapprox, 2024-11-14:
i experience still the laser pitch shift delay with the SYNC IN clocked with a very slow clock ONLY, the clock used here , is one dot at the begenning of the loop of 64 step Beatstep pro sequence.

hirnlego, 2024-11-14:
> i experience still the laser pitch shift delay with the SYNC IN clocked with a very slow clock ONLY, the clock used here , is one dot at the begenning of the loop of 64 step Beatstep pro sequence. i feel now the Echo more glitchy to different set of clocks .. and only stable and without pitch shifting in 1 bar loop clock.. i maybe suggest to simply remove the routing of the delay to the sync IN , ( triggering the loop for me is more crucial than having a divider in the delay ..

I don't know what BPM are you sending to the module, but as I said in the other ticket, if it's too slow (< 30 BPM) the synchronization will not work correctly and you should expect glitches.
I tried again with a 30 BPM sync with no problem whatsoever, so for me the issue is resolved.

hirnlego, 2024-11-14:
> Reporting back after testing the fresh new beta 7 - to be honest did not seem much different from beta 6.
> 
> > Please refer to my comment yesterday for details on the test scenario : [#22 (comment)](https://github.com/Befaco/Oneiroi/issues/22#issuecomment-2471695244)
> 
> Had a few loud non-repeating clicks as I was changing the values of LENGTH & VARISPEED (same as beta 6). And like with beta 6, I ended up finding spots on these 2 axes of looper manipulation (LENGTH & VARISPEED) where the repeating looping clicks appear - guess that makes me a a bit of a clicks digger 😅
> 
> And I tested with PRE (recording the output of an external wavetable VCO) as you suggested with the same results.
> 
> An interesting find (cannot say if it's specific to beta 7 vs 6, I'd bet it's not) - 3PM on both params seems like a gold mine for those damn clicks !

Ok, I confirm that there's a bug in the write fade in/out, I'm working on it!

hirnlego, 2024-11-14:
> Can you confirm this is how it's supposed to operate:
> 
> Start knob: from 0-100% the start point should move forward through the sample as you turn, end point remains static Length knob: at 100% is the full length of the sample, turning counterclockwise towards zero shortens the loop until it meets the start point, wherever that may be.

Yes, this is how START and LENGTH are supposed to work, I fear that something's not working right with your unit.
Try to recalibrate as you suggested and let us know.

hirnlego, 2024-11-16:
Beta 8 is up: https://github.com/Befaco/Oneiroi/issues/25

@cdromain please check if the click situation has improved 

Closing this one, see you in the other.

closed 2024-11-16 by hirnlego