Befaco/Oneiroi #25: Beta 8 for v1.1

hirnlego, 2024-11-16:
Here:

https://github.com/Befaco/Oneiroi/releases/tag/v1.1-beta.8

cdromain, 2024-11-17:
Thanks @hirnlego for your continuous work, dedication and the new beta 8 that I was eager to try 🙌🏼

The startup LEDs and FX leveling are great additions !

Unfortunately, I didn't notice any improvement regarding the looper clicks, that I could reproduce quite easily tonight, especially focusing on that 3PM area on both VARISPEED and LENGTH (see my previous reports in #22 &  #24  for more details). Let me know how I can help further, almost there 🚀

hemmer, 2024-11-18:
@hirnlego - echoing cdromain's comments, great work there.

I too am having issues with the looper clicking still though, but in hardware and VCV versions. I'm just scrubbing through the loop by varying the start point slowly and hear clicks. You can see these in the waveforms below too.
 
*Hardware Oneiroi*
[image]
[image]

*VCV version*
[image]

hirnlego, 2024-11-18:
@hemmer @cdromain but are you experiencing clicks only while turning the knobs or even without touching them?

hemmer, 2024-11-18:
Yes, here is an example with knobs left static. Admittedly it's more of a problem audibly when changing them, but for me scrubbing the start position is a common perfomative guesture. 

[image]

hemmer, 2024-11-18:
Another example here: 

[image]

cdromain, 2024-11-18:
> @hemmer @cdromain but are you experiencing clicks only while turning the knobs or even without touching them?

Hey @hirnlego, all my recent tests focused on what I call "looping clicks", i.e. clicks occurring systematically on each loop once knobs have been left untouched (e.g. that 3pm area on VARISPEED & LENGTH) - please see my previous comments. Lemme know how I can help further 🙌🏼

hemmer, 2024-11-20:
hey @hirnlego, i've been carefully debugging through and have found a few things to the point where I can at least get VCV version sounding nice (don't know how to build the hardware patch version though!). 

1. it looks like occaisionally the sample immediately after the `loopCrossFade_` windowing ends sometimes clicks because `phase_` is not reset in time. Setting it equal to any phase remainder (`phase_ = loopCrossFadePhase_ - kLooperFadeSamples;)` on exit from the loop cross fade logic seems to fix this for me. (looks like this already done for the trigger fades, just not the loop fades)

2. I think it's still nice having a little windowing on shorter samples. I settled for 0.1 seconds of time if long, or 5% of the loop if short - this eliminates clicking at short (counter-clockwise) lengths

3. finally when scrubbing start position, i can remove the zippering effect by only picking up new start positions on the next loop.

This changes can be seen in the below commit, let me know if it is helpful!

https://github.com/Befaco/Oneiroi/commit/f66ef585a92effb0d461216d4c7d0e0bcdc08f84#diff-fb2b06a5b496436a5d5a35c69ae650333494c88c3ccd334dbab940407ad93ea3

hirnlego, 2024-11-21:
@hemmer great! I've implemented your suggestions and also removed the DC filter when writing in the buffer that was causing other clicks when stopping recording (the filtering is done upstream anyway).
I sometimes still get clicks when moving START or LENGTH with just a sine tone recorded, I'm not sure how to tackle those, sine waves are not to be messed with

hemmer, 2024-11-21:
> Or by crossfading the output of the old phase with that of the new one

Isn't that's what happening in my suggested change? It starts ramping up the upcoming new loop (refers to newStart_) and fades out the old loop. By the time the new upcoming loop is at full volume, phase_ will reflect where loopCrossfadePhase_ had got to no?

hirnlego, 2024-11-21:
Well, yes, it's an alternative version of what was implemented before the change I made, but in this case it only handles changes in start position and just with forward playback. It's ok but there are other cases to cover (length changes and backwards playback), so I was thinking that maybe the old implementation covered everything. 
Right now length changes are immediate and can cause phase to jump before the loop fade has a chance to begin.
In any case, I still can hear clicks with a simple sine wave when changing either start or length, so probably there's something else that still can be improved.

hemmer, 2024-11-21:
Ok yes good point, I had meant to cover length changes but forgot. And yes also about reverse!

hirnlego, 2024-11-22:
> Ok yes good point, I had meant to cover length changes but forgot. And yes also about reverse!

Anyway, I'm working on a complete solution.

markzegarelli, 2024-11-24:
Attempted the update today, but the module seems bricked after. No lights or activity during power-on, and can't connect to the OWL.

Are there any recovery steps available?

hirnlego, 2024-11-29:
The final release for v1.1 is ready, get it here: 

https://github.com/Befaco/Oneiroi/issues/26

I think that I've managed to fix every case of clicks with the looper that I came across, @cdromain please let me know!

closed 2024-11-29 by hirnlego