Befaco/Oneiroi #26: New version 1.1

hirnlego, 2024-11-29:
Finally v1.1 is ready, get it here:

https://github.com/Befaco/Oneiroi/releases/tag/v1.1

hemmer, 2024-11-29:
Amazing looking forward to trying out, just checking - StereoWavetableOscillator.h references `EnvelopeFollower.h` which I can't see in git. I can use "EnvFollower.h" and rename the class and arguments if needed, but if you have the source do you mind to push?

hirnlego, 2024-11-29:
@hemmer ops, that was the OWL library's envelope follower, good catch. Fixed!

cdromain, 2024-11-29:
Hey guys, just tested and it looks like the looper is broken, not recording anymore ?
Should I download again (think I downloaded before the oops above) ?

hirnlego, 2024-11-29:
@cdromain no problems on my end... Not recording meaning that the record button doesn't work or that the buffer isn't written? Try do download again, but shouldn't be related in any way...

cdromain, 2024-11-29:
Buffer not written yes (rec button lights up red just fine) - actually I got it written once but that's it.
Ok will download again and report back !

cdromain, 2024-11-29:
Installed the latest build and I'm still getting this erratic non-recording behavior (for the first time, never happened with previous betas) - like I get the buffer written once in a while. What could be going wrong ?

hirnlego, 2024-11-29:
@cdromain don't know what to say, I'm not having issues with recording. Let's see if @hemmer is experiencing the same problem, in the meantime I demoted this version to Release Candidate. Would you mind trying to find a pattern in that behavior?

hemmer, 2024-11-29:
I just managed to get VCV version building, the buffer is now not interleaved right (was LRLRLR, now LLLRRR), took me a little time catch that! I'll test hardware version tomorrow night and confirm!

cdromain, 2024-11-29:
Haha that's all I've been doing man, looking for a reproducible pattern !
Seems the looper records fine when playing (i.e. varispeed not at noon) but not when stopped.

I also realized that the echo is acting weird as well, like I didn't get any change when moving density... Then came back to normal.

hirnlego, 2024-11-29:
> I just managed to get VCV version building, the buffer is now not interleaved correct? (was LRLRLR, now LLLRRR), took me a little time catch that! I'll test hardware version tomorrow night and confirm!

Yes, sorry, I changed because I was having some problems with the interleaving reading. I hope it hasn't drive you too mad...

hirnlego, 2024-11-29:
> > @cdromain don't know what to say, I'm not having issues with recording. Let's see if @hemmer is experiencing the same problem, in the meantime I demoted this version to Release Candidate. Would you mind trying to find a pattern in that behavior?
> 
> That's all I've been doing man, looking for a reproducible pattern ! Seems the looper records fine when playing (i.e. varispeed not at noon) but not when stopped.
> 
> I also realized that the echo is acting weird as well, like I didn't get any change when moving density... Then came back to normal. There's clearly something not right with this release, at least on my Oneiroi.

It's like you flashed another version! I didn't even touch the delay :/

The other parts of the module are working right?

cdromain, 2024-11-29:
> > > @cdromain don't know what to say, I'm not having issues with recording. Let's see if @hemmer is experiencing the same problem, in the meantime I demoted this version to Release Candidate. Would you mind trying to find a pattern in that behavior?
> > 
> > 
> > That's all I've been doing man, looking for a reproducible pattern ! Seems the looper records fine when playing (i.e. varispeed not at noon) but not when stopped.
> > I also realized that the echo is acting weird as well, like I didn't get any change when moving density... Then came back to normal. There's clearly something not right with this release, at least on my Oneiroi.
> 
> It's like you flashed another version! I didn't even touch the delay :/
> 
> The other parts of the module are working right?
> 
> EDIT
> 
> Wait, you're right, the looper wasn't recording when stalled. Please download the patch again, I should have fixed it!

 I'm 100% positive I flashed the latest 1.1 : [101980][0xf270d968]
Yes the rest seems to be working fine as far as I can tell.

And I confirm the recording only when playing behavior.

Downloading again, will report back !

cdromain, 2024-11-29:
K the third one was the good one 😃👍🏼
Looper working as expected.
And it was worth it as I'm happy to report that the click stituation (looping clicks I could easily provoke around 3PM on VARISPEED & LENGTH) seems to have improved substantially - could only get loud but non-looping one-time clicks when moving length, and it's been rare. Kudos man 🙌🏼 
RC seems good to go 🚀
Love Oneiroi !

hirnlego, 2024-11-29:
@cdromain great, thanks a lot for your support! 🙏

hirnlego, 2024-11-30:
> hello ! there is a problem now with inverted CV for the V/oct , sending higher CV plays lower notes and viceversa , i can solve that with shit mod atm + Random and Record , but then when i want to save that as default nothing is saved , when i switch on /off the module ... i think this was solved until maybe beta 6 ( im not really sure when exactly ) but not it gone wrong again ..

Well, this was the problem with CV mapping configured as attenuverters, but it's fixed since the first beta and never touched again. Anyway, I went back and tested again both attenuators and attenuverters mode for CVs (specifically V/OCT) and everything works, saves and resets as expected...

cdromain, 2024-11-30:
Morning guys, I can confirm that the V/OCT input works fine here as well !

I noticed this morning that Oneiroi was only picking the CV pitch info from my Hermod+ starting from just above 0V (i.e. no note change below that). I guess it's to be expected as the manual states that :
> "the v/oct input is common for the three oscillators, is summed to the PITCH knob, and tracks 10 octaves on the 0-volt to 10-volt range." (page 11)

Maybe page 6 should be modifed as we can read there that :
> "V/Oct input. Range -5V to 10V"

hirnlego, 2024-11-30:
> Maybe page 6 should be modifed as we can read there that :
> 
> > "V/Oct input. Range -5V to 10V"

Will correct, thank you.

cdromain, 2024-11-30:
> hello ! there is a problem now with inverted CV for the V/oct , sending higher CV plays lower notes and viceversa , i can solve that with shit mod atm + Random and Record , but then when i want to [...]

Have you tried recalibrating your Oneiroi ?

rootapprox, 2024-11-30:
@cdromain should i need to calibrate each 0.1 beta release ?

hirnlego, 2024-11-30:
@rootapprox no, you shouldn't, but doesn't hurt to try. In any case, it doesn't seem to be related to this specific version of the patch.

hemmer, 2024-11-30:
If I randomize loads (x100), I very occasionally get freezes on VCV version, I assume same can happen on hardware. 

ReadLeft takes a uint32_t here (https://github.com/Befaco/Oneiroi/blob/main/LooperBuffer.h#L204) so can never correct for negative phases (and therefore positions). Instead the value underflows and the while loop needs to do a massive set of substractions. But the phase can be negative in my experience, and the value passed in has type int32_t (presumably in expectation of this)? https://github.com/Befaco/Oneiroi/blob/main/LooperBuffer.h#L239

hirnlego, 2024-11-30:
> If I randomize many times (x100), I very occasionally get freezes on VCV version, I assume same can happen on hardware.
> 
> ReadLeft takes a uint32_t here (https://github.com/Befaco/Oneiroi/blob/main/LooperBuffer.h#L204) so can never correct for negative phases (and therefore positions). Instead the value underflows and the while loop needs to do a massive set of substractions. But the phase can be negative in my experience, and the value passed in has type int32_t (presumably in expectation of this)? https://github.com/Befaco/Oneiroi/blob/main/LooperBuffer.h#L239

Yes, I forgot to change those two methods, thanks for pointing it out. Will fix as soon as possible.

hirnlego, 2024-12-01:
@hemmer fixed and updated the release binary.

hemmer, 2024-12-01:
> Yes, sorry, I changed because I was having some problems with the interleaving reading. I hope it hasn't drive you too mad...

I think the WaveTableBuffer is still assuming that the looper buffer is interleaved? https://github.com/Befaco/Oneiroi/blob/main/WaveTableBuffer.h#L59

hirnlego, 2024-12-01:
> I think the WaveTableBuffer is still assuming that the looper buffer is interleaved? https://github.com/Befaco/Oneiroi/blob/main/WaveTableBuffer.h#L59 (though I don't 100% know what p1,p2 are)!

You're thinking right, I totally forgot about that... Thanks a lot, fixed! :facepalm:

cdromain, 2024-12-02:
Hey guys, is 1.1 considered final now ?
Was waiting before installing it again.
If so we should spread the word on ModWiggler !

hirnlego, 2024-12-02:
> Hey guys, is 1.1 considered final now ? Was waiting before installing it again. If so we should spread the word on ModWiggler !

Yes, it's final! Go ahead :D

cdromain, 2024-12-02:
Awesome, thanks man - will do 🚀

5StringFiddler, 2024-12-27:
Hi all,   My Oneiroi  v/o is not calibrating V/O correctly- the octaves is about a semitone off at one octave., ( I have tried to calibrate several times, with different voltage sources)  Is there a way to directly edit the calibration files, as opposed to using the 2V, 5V 8V method?

5StringFiddler, 2024-12-27:
One other quick question-   the VCA speed seems to be slewed quite extreme to about .12 seconds when feeding a gate (either on or off from a 10 volt gate)   When looking at the schematic, it seems to be a software limitation and not a hardware limitation.    It would be nice to have faster gates on the VOL CVs to respond to control voltage.   Is that possible to address in a future build, or add as an option?
attached is a pic  comparing the gate response of my Pony VCO's VCA  and the same gate fed to Oneiroi's Saw VCO. 

[image]

Lemmike, 2025-01-04:
> Hi all, My Oneiroi v/o is not calibrating V/O correctly- the octaves is about a semitone off at one octave., ( I have tried to calibrate several times, with different voltage sources) Is there a way to directly edit the calibration files, as opposed to using the 2V, 5V 8V method?

I am having the same issue. Pitch tracking is off and I also have constant pitch jitter when using the V/OCT input. Also I am unable to calibrate the pitch knob to a clean C in the middle position even after calibrating several times.

5StringFiddler, 2025-01-04:
I do have that pitch wobble that you are describing as well.

GoodFellow-801, 2025-01-23:
Having same issue reported by @cdromain .  Just updated to v1.1 and recording to the looper is not functioning correctly.  When recording anything (supersaw or external source via inputs) all I ever hear in the looper is what sounds like a triangle wave.  It responds to verispeed pitch changes, but resembles nothing of the source audio.  Triangle wave sound is much quieter when recording with switch set to "pre".  I had none of these issues with v1.0.

heyx3, 2025-03-03:
I too have had issues playing with the recording and Looper, at least in the VCV version of this module.

Diegoterrores, 2025-03-13:
Closed for clarity as this one contains too many different issues in one
If you encounter a problem, please check if your issue fits the existing ones. If not open a new one

closed 2025-03-13 by Diegoterrores

AuAuAudio, 2025-04-03:
Hello!

Having a white noise glitch when I start a loop, then after manipulating it randomly or recording on top the white noise disappears, but this makes it unusable in many occasions and not trustable to use live. 
Is the Eurorack DIY version and everything else works perfectly.
Any ideas on how to solve it? could it be a malfunctioning opamp?
Many thanks,
Pedro

Diegoterrores, 2025-04-03:
By desing, the buffer is filled with white noise at start-up and and you
record the recording override the noise this what you mean by a chance?
If this is the case SHift+ record erase the buffer ans the noise

AuAuAudio, 2025-04-03:
that's exactly it! shift + record solves the problem. If I record in the beginning and don't fill a full loop, there would be parts of white noise in the loop. But I can hit shift + rec every time before making the 1st loop and I'm ok.
Many thanks for the explanation!
thought it could be my unit bad behavior, has I did it DIY...

Diegoterrores, 2025-04-05:
Glad to hear it!

hansiwalker, 2025-04-24:
Hi, yesterday I installed version 1.1. This seemed to fix the 'inversed pitch cv issue'. Today pitch cv is inverted again though at startup the blinking LEDs say version 1.1.

Any help appreciated,
Hans

chrisib, 2025-05-29:
I'm a little late to the party, but I'm trying to update my Oneiroi to 1.1 right now. After I click the `store` button I see the loading percentage increasing, but it never seems to reach 100. It stops somewhere between 80-97% (usually at 95).

I've tried two different browsers (Firefox & Edge), I've tried clearing the cache, I've tried reloading the page, I've tried just leaving it alone for 30 minutes to see if that final few percent just take forever, and I've tried power-cycling the module. No matter what, it just never seems to reach 100 and the webpage says "Connected to Oneiroi v1.0."

Is there some trick to getting https://pingdynasty.github.io/OwlWebControl/extended.html to actually finish storing the file? Is this indicative of a build error?

cdromain, 2025-05-30:
Hey, the "connected to 1.0" is to be expected - look for the end value of the line "Preset 1: Oneiroi [102044][0xffc55af8]" the end part should change after a successful update.
In my experience I was reaching the "Program Message: Loading 100" mark though. Have you tried with Chrome as well ?
Good luck !

scottpig, 2025-06-22:
Where is the serial number shown, specifically on DIY kits? I don't recall seeing one anywhere and I can't see one on it now that it's assembled. I'm trying to resolve the inverse-pitch tracking issue.