Befaco/Oneiroi #27: CV inversion after saving to flash .

rootapprox, 2024-12-01:
> the only way to solve it is to Save my parameters then open the resources files and delete the cv mod file , then it comes back again normally ..

That's exactly what I was suggesting, to erase the file. If you still have issue with that, please open a ticket, thank you!

klangmuster, 2024-12-03:
have a similar issue. not only v/oct is affected but also other cv modulation e.g. cv input for filter frequency.
with no ressources except of the oneiroi.cfg everything is fine.
as soon as i save (long press mod amt) aftr the next power cycle the cv input in inverted.
have either to delete some ressource files or click (mod amt - shift + record - random) to get the normal cv mapping back.
super annoying...

klangmuster, 2024-12-04:
i think i found the issue. seems sth is not ok during the calibration process.
seems the cv range is setup wrong during calibration (as described in the manual).
if you go into cv mapping mode after the calibration and set all knobs to fully cw and save these settings everything afterwards works as expected. any assignment is correct and also works as expected after the next power off/on.

closed 2024-12-09 by hirnlego