Befaco/Oneiroi #28: Issues with mod assignments and 1v/oct

blave, 2024-12-06:
I just installed my Oneiroi into my brand new Cre8audio Niftykeyz keyboard/skiff. I had it in my 2x150 Goike skiff before that and it seemed to work OK, although sometimes I would have to re-power to get it to work right (when malfunctioning, only the level LED would be lit, dimly). 

The module hasn't done that in the new skiff, but I am having two other issues. 

1. I can't seem to disable modulation of anything using the MOD AMT button. Right now the pitch, filter, and resonator are modulated by the LFO and I can't disable those mappings, by hitting the MOD AMT button and then turning each knob all the way to the left. I've tried deleting the mappings to no avail. It did this earlier today, and so I (think I) upgraded to 1.1 (I got what appear to be the correct log messages during the update). It's still doing it. The modulation mapping worked correctly a few days ago.
2.  I have the keyboard's CV1 1v/oct output patched to the Pitch knob. When I do that, the pitch emitted is at a very low, almost sub-sonic frequency, and neither the keyboard or the pitch knob changes the pitch. If I unplug the pitch CV, the pitch knob behaves normally.

Now, I have never been sure what firmware my Oneiroi shipped with. It's from a brand new batch that Detroit Modular got (it was backordered when I ordered it) about two weeks ago, so am I running into the issue where if I have 1.0 I have to use calibrated voltage sources to get things right? I can do that, but it would be kind of a PITA (my adjustable power supply is in storage) so I've avoided it.

Ideas/suggestions?

blave, 2024-12-06:
BTW I have verified that the keyboard's 1v/oct works correctly, with an MI Braids.

hirnlego, 2024-12-06:
@blave when installing v1.1 did you erase the configuration files as specified in the release notes? In the same release notes there's also the explanation on how to check for the release version using the module's LEDs.

blave, 2024-12-06:
I did not. I thought that only had to be done for upgrading from the original firmware version. And also, I totally missed the note about the LEDs. Thanks for your response! I'll dig into this tomorrow. I thought I'd read everything, but... Apparently nope!

blave, 2024-12-07:
OK deleting those files and re-flashing 1.1 greatly improved things. Thanks for the guidance!

hirnlego, 2024-12-07:
Great, glad it worked!

closed 2024-12-07 by hirnlego

blave, 2024-12-08:
Not sure how to re-open an Issue, but I am still experiencing this:

"I have the keyboard's CV1 1v/oct output patched to the Pitch knob. When I do that, the pitch emitted is at a very low, almost sub-sonic frequency, and neither the keyboard or the pitch knob changes the pitch. If I unplug the pitch CV, the pitch knob behaves normally."