Befaco/Oneiroi #30: Start/Length knobs stop functioning

blave, 2024-12-08:
I'm not sure what sequence of actions gets me here, but this has happened twice today. The start and length knobs will work fine, until they suddenly become totally unresponsive for some reason.. And the 2nd time, the looper pitch control also stopped working. Re-powering clears the problem. For a while I thought it was after I had used the (most excellent) Randomizer function, but I was not able to reproduce it just now by repeatedly pressing the Random button..

hirnlego, 2024-12-09:
@blave let us know if you're able to reproduce it consistently, thanks!

blave, 2024-12-09:
Will do.

Diegoterrores, 2025-03-13:
This issue is being closed due to inactivity. 
Please reopen it if needed.

closed 2025-03-13 by Diegoterrores