Befaco/Oneiroi #33: CV Controll

Beatalchemie, 2024-12-18:
Hi
The CV inputs for the sliders (for VCOs & DRY/WET) are designed for 10 volts. Most modulation sources (e.g. ADSR, uni LFOs etc.) are 5 volts. And so the modulation level is not enough to fully open the sliders and that is a problem.
Is it possible to make the CV inputs unipolar at 5 volts or -5 to +5 volts and declare the controller as an attenu value?
That would be really fantastic.

And I have already said that the [WT] VCO is always an octave higher than the SINE VCO or SS, which is really annoying!

Otherwise, thank you very much for the wonderful module. I always use it now, with every patch hehehe. Totally cool

markcrosbie, 2025-01-19:
Yes I noticed this as well today when shooting a video - and even if an envelope gets to 10v the result isn't as loud as manually opening the slider

Diegoterrores, 2025-03-05:
The faders are designed to be "normalized" to the control voltage (CV) inputs, which means that bipolar control is not possible. Instead of adding an offset to the CV, the faders function as attenuators, meaning that negative voltage will have no effect.

One potential improvement could be to adjust the range to a smaller positive value, such as 5 volts. 
While it is possible to change the range to 0 to 5 volts, doing so would require hardware modifications. If user demand warrants it, this change could be implemented in a future hardware revision.

Diegoterrores, 2025-03-11:
It may be beneficial for the Vots vs. Volume ratio to exhibit a more exponential response. We could explore the possibility of finding a middle ground where most of the volume is achieved at 5 volts. This improvement might be attainable through a software update. 

@hirnlego 
Can we conduct some research on this?

Pa-Ulta, 2025-03-23:
Would it be possible, perhaps, to have a startup/config option to offset the incoming voltage globally for all fader cvs, so -5v to +5v becomes 0v to 10v? I can, of course, do this with some attenuverters, but that would require multiple attenuverters.

Diegoterrores, 2025-03-25:
Not possible, as the current hardware does not accept negative voltage at the volume INs.