Befaco/Oneiroi #38: Resonator emitting high frequency ringing

geraasmasjien, 2025-03-01:
On the Resonator there seems to be a constant ringing at roughly 8200Hz and 16400Hz (these are not accurate measurements but should be very visible on any spectrum analyser. 

It doesn't seem to respond to the tuning of the Resonator which leads me to believe that it isn't part of the features of the Resonator. 

Would it be possible to filter these our somewhere in the code since it detracts from the character of the Resonator. Perhaps it is caused by aliasing?

Diegoterrores, 2025-03-05:
It appears to be some parasitic oscillation along with its second harmonic. We will investigate that further. 
Thank you for pointing it out!

Diegoterrores, 2025-03-12:
I have tested the setup, but unfortunately, the noise issue still persists (see graphs). It appears that the noise is proportional to the input volume, which suggests it might be related to aliasing. 

I also noticed that the parasite frequency is at around 8 kHz, and the 16 kHz frequency is likely a harmonic of it. This frequency seems to correspond to 1/6 of the sampling frequency, which is 48 kHz, along with the default block size of 128 samples (as I believe). 

Can someone confirm if this is accurate in the Oneiroi firmware? It might provide us with a useful clue.

[image]
[image]

hirnlego, 2025-03-12:
@Diegoterrores block size is actually 32. In your graphs I see the dips caused by the notch filters, it doesn't appear to be a good solution. I'll make more tests.

@hemmer do you have the same issue in the VCV version?

Diegoterrores, 2025-03-12:
I can confirm that the same artifact is audible in the software version. 
With only the sine wave at maximum and the resonator set to wet, it is very clear, especially when the resonator's tuning pot is past 12:00.

Diegoterrores, 2025-03-12:
Sounds like aliasing to me

hemmer, 2025-03-12:
It's interesting! In VCV you can oversample and it's mitigated (amplitude of parasitic peaks gets lower, but not eradicated). It also seems to consistently occur at sampleRate/6 (+ harmonics above that), for any sample rate I try.

It's aliasing-like, but somehow with aliasing I wouldn't expect **fixed** peaks (i.e I would expect material above Nyquist to be reflected back, but changing the resonantor tuning would change where they are reflected back to no?)

hemmer, 2025-03-12:
Also when I recompile with different block rate (VCV again), the peaks don't change, so not to do with block processing.

hirnlego, 2025-09-22:
> I received my Unit yesterday and confirm this bug is happening for my Oneiroi as well. Is this something that can be fixed through FW or is this a HW related issue ? based on [@hemmer](https://github.com/hemmer) `s results with the VCV rack version I assume this is FW related?

It's not an HW issue, so it might be resolved in a future patch update.

closed 2025-12-02 by hirnlego