Befaco/Oneiroi #45: V/oct not working

dgbc, 2025-05-27:
Hi.

I'm experiencing the same issue as described in https://github.com/Befaco/Oneiroi/issues/43
Is there another firmware after 1.1 that can fix it?

Thanks

djfresha, 2025-06-05:
Same issue here. I try to calibrate and when it goes through it seems all ok, but then I get a super super low sine and a ridiculously high saw/supesaw (dog whistle territory) and V/Oct won't retune properly.

Just did a firmware blank (patch.syx) in boths slots 0 and 1, restarted the device and now got the standard pitch right but no V/Oct

dgbc, 2025-06-06:
There's no Makefile in the repository, which means that we can't use the make commands in the README to compile it ourselves.

djfresha, 2025-06-06:
Yeah that's what I'm considering, plus that link you provided looks like it goes to v2? has makefilein there too.

https://github.com/RebelTechnology/OpenWare/tree/feature/oneiroi/Oneiroi

Also waiting on Befaco to see what they are saying

dgbc, 2025-06-06:
> Yeah that's what I'm considering, plus that link you provided looks like it goes to v2? has makefilein there too.
> 
> https://github.com/RebelTechnology/OpenWare/tree/feature/oneiroi/Oneiroi
> 
> Also waiting on Befaco to see what they are saying

The code on OpenWare's repo looks like an old version of oneiroi, you can see some similar structs that have been move to different .h files, instead of the big .cpp files.

djfresha, 2025-07-03:
Quick update guys - firmware and patch 1.2 has been launched, and it's actually got my Oneiroi sorted so the V/Oct socket is reading correctly. 

There's a firmware Openware page to go to first when doing the firmware specifically, not the same as the one where we load the patch in. Once firmware is done you may get the unit all red LEDs on 'locked up' and your browser might say 'invalid store command' or similar.  Ignore that message, turn off your Eurorack and then back on, then you can load up the v1.2 patch via the 'human way' link. If you want to be sure you have got the right firmware loaded, the RebelTech connect to Owl page will tell you the firmware - if v1 / v1.1 you'll see just Oneiroi, v1.2 displays Oneiroi_v1_2 (iirc or along those lines)

Once this firmware and patch loaded I could do V/Oct pitch controls like I would expect, advise doing the calibration but a bit easier now as we only need to setup the pitch knob and the looper knob.

Good times arolling :D