Befaco/Oneiroi #49: Can't update to 1.2

boorch, 2025-07-16:
I followed instructions, while installing firmware 1.2, i receive this log:

Reading file Oniroi_firmware-v1.2.syx
Sending data 365422 bytes
Program Message: Loading 0
Program Message: Loading 15
Program Message: Loading 26
Program Message: Loading 29
Program Message: Loading 44
Program Message: Loading 55
Program Message: Loading 58
Program Message: Loading 73
Program Message: Loading 84
Program Message: Loading 87
Program Message: Loading 100
Error: Error 0x60 Invalid bootloader
Error: Error 0x60 Invalid bootloader
Error: Error 0x60 Invalid bootloader
Error: Error 0x60 Invalid bootloader
Error: Error 0x60 Invalid bootloader
Error: Error 0x60 Invalid bootloader
...

when i try to "reset" device, i see an error like "Error 0x70 Preventing reset program from starting"

also i noticed in the code, Patch minor version is still 1, so even if i update somehow, random led blinks once as if it's 1.1

blindorange, 2025-07-24:
I have the same error, I deleted the .cfg file in resources inside [https://www.rebeltech.org/patch-library/device/] opened [https://pingdynasty.github.io/OwlWebControl/firmware.html] sent the patch file in [https://pingdynasty.github.io/OwlWebControl/extended.html] then sent the firmware in [https://pingdynasty.github.io/OwlWebControl/extended.html] and saw the same message Error: Error 0x60 Invalid bootloader.

The moduke freezes at this point with all lights lit

blindorange, 2025-07-24:
Update, working; I disconnected the power, changed to another USB cable, went here [https://pingdynasty.github.io/OwlWebControl/extended.html] chose the firmware, hit send and the update completed, though the firmware is now noted as just Oneiroi with no 1.2 reference which is confusing. The patch states Oneiroi_1_2_0