Befaco/Oneiroi #50: Weird behavior of looper after some presses of random button

BefacoSupport, 2025-07-22:
The looper, start and length knobs have a weird behavior after pressing the Random button a few times. The start and length seem to stop working.

hermansmasher, 2025-08-06:
> [@BefacoSupport](https://github.com/BefacoSupport)
> 
> > this seem not to happen, in firmware version 1
> 
> Do you mean 1.0 or 1.1? Because we haven't touch anything related to the looper or the random function in 1.2 (I'm assuming that this issue has been reported while running 1.2).

Hey :)
First, sorry for spamming the release tab, I didn't know.
Second, correlating with the date of posting, I guess this ticket was opened due to my mail sent to Befaco-Support (22nd July). I was first in contact with Mattin and then with Andrés, who confirmed being able to recplicate this bug.
Just came here to answer your question that I was indeed running v1.2 and can reliably recreate this "stuck Start and Length" behavior on my module. Also another user on Modwiggler reported the very same problem on v1.2

hermansmasher, 2025-08-27:
> So, I've got a chance to put some time in this issue but I'm afraid I cannot reproduce it by smashing the random button, Start and Length keep working as expected. In any case, it's weird that only these two knobs should be affected, it doesn't make any sense. Moreover, I haven't touch anything looper-wise going from 1.1 to 1.2, so this problem appearing now in this release is mysterious.
> 
> I need more info, or an exact pattern I can follow to replicate the problem, otherwise I'm stuck.

As another modwiggler-member wrote "Yep, hit this one yesterday. It seems hard to reproduce cause it might work fine for 30+ random button presses, switch the module off-on again and will happen after 10 quick presses…"

It seems to mainly occur on quick presses and when start and length is not touched in between. As mentioned in the quote above, I also can't tell how many button presses are needed for this to happen, seems different everytime ..

hirnlego, 2025-09-14:
> As another modwiggler-member wrote "Yep, hit this one yesterday. It seems hard to reproduce cause it might work fine for 30+ random button presses, switch the module off-on again and will happen [...]

Unfortunately I'm still unable to reproduce it, so the next best thing for me is to better understand what happens to the looper when this issue presents itself. A few questions for you:

- what does exactly happen to the start and length knobs? In what sense they "stop working"?
- are you able to change start and length with CVs when the knobs stop working?
- what happens to the other looper controls? Do they still work?
- and what about the rest of the module? Still working as expected?
- do further randomizations resolve the issue for the start and length knobs? And what about the rest of the looper parameters, do they still change as expected?

Thank you!